pilsbot/pilsbot #2: Choose steering-type

Cirromulus, 2019-08-12:
# Request for comments

## Hardware Ackermann steering (Hackermann)
As seen in preview
Pro:
- Only one servo needed
- Mis-configuration of wheels ruled out

Con:
- Fairly complex hardware construction

## Software Ackermann steering (Sackermann)
Two wheels, two independend servo motors
Pro:
- Easier hardware layout
- Interesting steering modes supported

Con:
- Two servos = more power consumption
- More complex software

ChrisStil, 2019-08-13:
I find Sackermann better because it's easier to build. If you put the two wheel in the middle of the construction, it would be even easier to control

PS: are you sure that Sackerman has a bigger power consumption? You still need an other motor ber Hackermann to throttle. With Sackerman, you could use both motor to throttle no?

Cirromulus, 2019-08-13:
> With Sackerman, you could use both motor to throttle no?
This is not quite right as we are using a single rear motor for throttle. The front axle is intended to be passive, with only steering purposes. 
But you are right, Sackermann might be adequate for our purposes.

Cirromulus, 2019-08-14:
Maybe, we just mis-use an existing lorry (e.g. [this from ebay](https://www.ebay.de/itm/Kesser-Bollerwagen-550kg-Transportwagen-Gartenwagen-Geratewagen-Handwagen-Power/222868790222)) and put some actuators on it.

ChrisStil, 2019-08-14:
If you want to put some actuators on it, maybe it would be better to buy a wooden lorry (big enough to put your speakers in it too)... Or build it

Cirromulus, 2019-09-13:
https://www.drive-system.com/elektrozylinder-dszy1-encoder.php
Netter 200er :(

Cirromulus, 2019-09-26:
https://www.motioncontrolproducts.co.uk/products/156/165/gd-linear-actuators/
This one is cheaper.

closed 2020-05-28 by Cirromulus