pilsbot/pilsbot #3: Aquire Motor & other components

Cirromulus, 2019-08-14:


ChrisStil, 2019-08-14:
Will we use a raspberry pi? If yes, I have a camera that can be inserted in the extra slot (but I have no avaiable raspberry Pi a.t.m.)

Cirromulus, 2019-08-14:
This may be too weak in computing power, but maybe our software expert may answer this @DarthBubi

DarthBubi, 2019-08-15:
Even a Raspberry Pi  4 is too weak in my eyes. If we want to do vision-based tasks, we better go for a [Nvidia Jetson Nano](https://www.mouser.de/datasheet/2/813/jetson-nano-devkit-datasheet-1545728.pdf). Costs about 110 € atm and has more compute power.

Cirromulus, 2019-08-15:
I think, this would be appropriate.
btw: This board seems pretty nice for this price.

Cirromulus, 2020-05-30:
- raspberry camera (Johas 70°)
- raspberry (hat Manu)
- logitech shitcam (PP temporär)
- 1-2 Arduino Pro Micro (erst mal von PP)
- 4-6 Ultraschallsensoren for Obstacle Detection (Joha kauft)
- 2 IR TOF for Stufe detection (Später dann)
- ? Simple laserscanner on bottom for more elegant Obstacle D

DarthBubi, 2020-07-20:
I'll compile a list of components for replacing the Raspberry Pi 3 with a Jetson Nano this week.