pilsbot/pilsbot #14: Create a Smartphone App for communication

PimpMyPizza, 2020-05-31:
I'll make a smartphone app ( [With flutter](https://flutter.dev/), so for IOS and Android) in order to communicate with the pilsbot over bluetooth. 
Features:
- Control the Robot manually
- Insert GPS-Waypoints

If you have more Ideas, put them here please <3

Cirromulus, 2020-06-01:
Would you include it into the main repo as a submodule, please?

Cirromulus, 2020-06-01:
A suggestion would be a status-page, where you can see all relevant info like
- Battery voltage (percentage)
- perhaps CPU-time
- Some kind of QOS (is this possible with detection errors or similar?)
- Perception of outside world (pointcloud, sensor data)

And a debug-page where you can tune all the parameters.

ps.: Thanks :)