pilsbot/pilsbot #25: Automatic installation of dependencies

PimpMyPizza, 2021-02-05:
- We use several tools/ROS packages in the project
- Some of these tools have dependencies (for example ros2-web-bridge depends on nodejs, that depends on nvm under ubuntu. The project in a whole depends on ros2 and vcstool)
- It would be cool to automatically install the dependencies depending on the OS used in order to avoid 10000 installation tutorials
- I'm not quite sure which solution is the best to do that.

Cirromulus, 2021-02-06:
I am not sure about that. I usually go with an installation script, but these tend to age unwell.
In the current state of "rapid" development, maybe one installs the packages on-the-go as errors come.

PimpMyPizza, 2021-03-02:
rel: #26

closed 2021-03-02 by PimpMyPizza