pilsbot/pilsbot #26: Setup bootstrap and reorganize repositories

hemker, 2021-02-14:


PimpMyPizza, 2021-02-14:
For the moment, you can download the repositories directly with `pilsbot update`. See the README in https://github.com/pilsbot/workspace . There is no skript for installing the deps atm and the scripts are rudimentary and only tested in ubuntu.

PimpMyPizza, 2021-03-01:
Maybe we should remove the `pilsbot_` prefix in the packages names

DarthBubi, 2021-03-01:
Why should we drop that? It is common practice to prefix your device-specific packages.

PimpMyPizza, 2021-03-02:
Okay, I just though that since all the packages are in the pilsbot group, the `pilsbot_` prefix was a little bit too much, but if that's a common practice, forget what I wrote. It would be cool if we use this for the related packages (some doesn't have the prefix even if they should, some have just `pils_` etc.

Cirromulus, 2021-03-02:
Lul

Cirromulus, 2021-11-29:
at least working demo is found under this example  workspace: [pilsbot/workspace](https://github.com/pilsbot/workspace)

closed 2021-11-29 by Cirromulus